GradientSpaces/LoopSplat #1: Missing requirements.txt in Hierarchical-Localization

abhishek47kashyap, 2024-08-21:
After setting up the conda requirement with all dependencies specified in `environment.yml`, I tried to follow the instructions to install _hloc_, however `python -m pip install -e .` errors out:
```
FileNotFoundError: [Errno 2] No such file or directory: '/home/abhishek/Code/LoopSplat/thridparty/Hierarchical-Localization/requirements.txt'
```
Seems like `requirements.txt` is absent?

Also, a really minor typo `thrid` in the directory name `thridparty/Hierarchical-Localization` :)

unique1i, 2024-08-21:
Hi @abhishek47kashyap, thanks for your spot! I uploaded the missing requirements.txt and renamed the folder.

Zhu-Liyuan, 2024-08-21:
I have fixed the submodule problem. Now it's working on my machine. I close this issue for now.

closed 2024-08-21 by Zhu-Liyuan