GradientSpaces/LoopSplat #2: Question about the visualization

xuziheng0310, 2024-08-23:
Hi, thanks for your great work! I'm interested in how you visualize the scene like Figure 1 and 4 in the paper. It seems your results differ from SuperSplat's outputs(if I'm using it correctly). Looking forward to your reply!

Zhu-Liyuan, 2024-08-23:
Thanks for your interest in our work. If I understood correctly, you run LoopSplat and visualize the 3DGS reconstruction in SuperSplat's tool. In figure 1 and 4, we render the colored images from mesh as they can better show the qualitative difference in both geometry and color.

closed 2024-08-23 by Zhu-Liyuan

xuziheng0310, 2024-08-24:
Thanks for your quick reply! So how do you render the mesh, is there a visualizer code?

Zhu-Liyuan, 2024-08-24:
We don't provide code for rendering mesh. There are many resources you can refer to like pyrender, open3d and blender.