GradientSpaces/LoopSplat #3: Failed to converge

yaorz97, 2024-09-03:
Thank you for your excellent work! When I try to reproduce the results on ScanNet, an error occurs. Can you tell me how to fix it? 

[image]

Zhu-Liyuan, 2024-09-03:
Thank you for trying our code. This is the first time I have encountered this, which never happens on my machine. It seems that the averaging produced some unexpected values.

You can add this in solver.py 
```
    try:
        R_w = roma.special_procrustes(M)
    except Exception as e:
        print(f"Error in roma.special_procrustes: {e}")
        return {
            'successful': False,
            "pred_tsfm": torch.eye(4).cuda(),
            "gt_tsfm": torch.eye(4).cuda(),
            "overlap": init_overlap.item()
        }
```
You can pull the updated code and try again.

Zhu-Liyuan, 2024-09-03:
Closing this issue. Feel free to ask again if you have any problems.

closed 2024-09-03 by Zhu-Liyuan

yaorz97, 2024-09-03:
Thanks for your prompt reply! I will try it again and compare the results with those reported in the paper.

Zhu-Liyuan, 2024-09-03:
Sounds good. Did you meet this bug at the beginning of the run or just on a single scene?

unique1i, 2024-09-03:
Hi @yaorz97, to double check, are the frames with inf values in poses filtered out? There are inf values in scene 0054 and 0059.

yaorz97, 2024-09-03:
Yes, I have processed the frames by `scripts/scannet_preprocess.ipynb`,  and the valid frames were selected. The gt_pose.txt file of scene 54 was saved and you can check it.
[gt_pose.txt](https://github.com/user-attachments/files/16850618/gt_pose.txt)