GradientSpaces/LoopSplat #5: Question About self.loop_closer.loop_closure

Tokyo-dream, 2024-09-04:
Traceback (most recent call last):
  File "/home/private/wangpengyu/LoopSplat/run_slam.py", line 109, in <module>
    gslam.run()
  File "/home/private/wangpengyu/LoopSplat/src/entities/gaussian_slam.py", line 238, in run
    lc_output = self.loop_closer.loop_closure(self.estimated_c2ws)
  File "/home/private/wangpengyu/LoopSplat/src/entities/lc.py", line 282, in loop_closure
    pose_graph, odometry_edges, loop_edges = self.construct_pose_graph(final)
  File "/home/private/wangpengyu/LoopSplat/src/entities/lc.py", line 237, in construct_pose_graph
    reg_dict = self.pairwise_registration(submap_list[source_id], submap_list[target_id], "gs_reg")
  File "/home/private/wangpengyu/LoopSplat/src/entities/lc.py", line 529, in pairwise_registration
    res = gs_reg(submap_source, submap_target, self.config['lc']['registration'])
  File "/home/private/wangpengyu/LoopSplat/src/gsr/solver.py", line 113, in gaussian_registration
    init_overlap = compute_overlap_gaussians(src_dict['gaussians'], tgt_dict['gaussians'], 0.1)
  File "/home/private/wangpengyu/LoopSplat/src/gsr/overlap.py", line 53, in compute_overlap_gaussians
    gpu_index.add(tgt_tensor)
  File "/root/miniconda3/envs/loop_splat/lib/python3.10/site-packages/faiss/contrib/torch_utils.py", line 122, in torch_replacement_add
    assert hasattr(self, 'getDevice'), 'GPU tensor on CPU index not allowed'
AssertionError: GPU tensor on CPU index not allowed

Tokyo-dream, 2024-09-04:
Thanks for this excellent work. When I run all the scenes, errors are reported. May I ask why this is?

Zhu-Liyuan, 2024-09-04:
Thank you for report the bug. It seems to be something wrong with faiss. I could not reproduce the bug on my local machine.
Can you try `gpu_index.add(tgt_tensor.cuda())` to replace this [line](https://github.com/GradientSpaces/LoopSplat/blob/77ab521b06bf74ec06e7efffd2a0b441d5b0b532/src/gsr/overlap.py#L53C19-L53C29)?

closed 2024-09-06 by Zhu-Liyuan

yaorz97, 2024-09-10:
I have also experienced this issue on a single RTX 3090.  It is finally addressed by replacing `gpu_index = faiss.index_cpu_to_all_gpus(cpu_index)` with `gpu_index = faiss.index_cpu_to_gpu(faiss.StandardGpuResources(), 0, cpu_index)`. You can try it on your machine.

Zhu-Liyuan, 2024-09-10:
@yaorz97 thank you for keeping updated! Did you manage to reproduce the results mentioned in [issue3](https://github.com/GradientSpaces/LoopSplat/issues/3)? If so, I can update the codebase with your solution. Thanks!