GradientSpaces/LoopSplat #6: fail to build in simple_knn , gaussian_rasterizer,  diff_gaussian_rasterization

jinxxo-j, 2024-09-23:
Building wheel for simple_knn (setup.py): started
  Building wheel for simple_knn (setup.py): finished with status 'error'
  Running setup.py clean for simple_knn
  Building wheel for gaussian_rasterizer (setup.py): started
  Building wheel for gaussian_rasterizer (setup.py): finished with status 'error'
  Running setup.py clean for gaussian_rasterizer
  Building wheel for diff_gaussian_rasterization (setup.py): started
  Building wheel for diff_gaussian_rasterization (setup.py): finished with status 'error'
  Running setup.py clean for diff_gaussian_rasterization
Successfully built evaluate_3d_reconstruction
Failed to build simple_knn gaussian_rasterizer diff_gaussian_rasterization
Pip subprocess error:
  Running command git clone --filter=blob:none --quiet https://github.com/eriksandstroem/evaluate_3d_rec
onstruction_lib.git /tmp/pip-req-build-12ulqv9i
  Running command git rev-parse -q --verify 'sha^9b3cc08be5440db9c375cc21e3bd65bb4a337db7'
  Running command git fetch -q https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git 9b3c
c08be5440db9c375cc21e3bd65bb4a337db7
  Running command git clone --filter=blob:none --quiet https://github.com/VladimirYugay/simple-knn.git /
tmp/pip-req-build-6hcn4vko
  Running command git rev-parse -q --verify 'sha^c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d'
  Running command git fetch -q https://github.com/VladimirYugay/simple-knn.git c7e51a06a4cd84c25e769fee2
9ab391fe5d5ff8d
  Running command git clone --filter=blob:none --quiet https://github.com/VladimirYugay/gaussian_rasteri
zer.git /tmp/pip-req-build-pnwv1oln
  Running command git rev-parse -q --verify 'sha^9c40173fcc8d9b16778a1a8040295bc2f9ebf129'
  Running command git fetch -q https://github.com/VladimirYugay/gaussian_rasterizer.git 9c40173fcc8d9b16
778a1a8040295bc2f9ebf129
  Running command git clone --filter=blob:none --quiet https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose.git /tmp/pip-req-build-o29ti5x6
  Running command git rev-parse -q --verify 'sha^43e21bff91cd24986ee3dd52fe0bb06952e50ec7'
  Running command git fetch -q https://github.com/rmurai0610/diff-gaussian-rasterization-w-pose.git 43e21bff91cd24986ee3dd52fe0bb06952e50ec7
  Running command git submodule update --init --recursive -q
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      running bdist_wheel
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:502: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_ext
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:414: UserWarning: The detected CUDA version (12.4) has a minor version mismatch with the version that was used to compile PyTorch (12.1). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:424: UserWarning: There are no /usr/bin/g++ version bounds defined for CUDA version 12.4
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'simple_knn._C' extension
      creating build
      creating build/temp.linux-x86_64-cpython-310
      /usr/bin/g++ -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/jinwoo/anaconda3/envs/loop_splat/include -fPIC -O2 -isystem /home/jinwoo/anaconda3/envs/loop_splat/include -fPIC -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/TH -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/THC -I/home/jinwoo/anaconda3/envs/loop_splat/include -I/home/jinwoo/anaconda3/envs/loop_splat/include/python3.10 -c ext.cpp -o build/temp.linux-x86_64-cpython-310/ext.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
      /home/jinwoo/anaconda3/envs/loop_splat/bin/nvcc -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/TH -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/THC -I/home/jinwoo/anaconda3/envs/loop_splat/include -I/home/jinwoo/anaconda3/envs/loop_splat/include/python3.10 -c simple_knn.cu -o build/temp.linux-x86_64-cpython-310/simple_knn.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -ccbin /usr/bin/gcc -std=c++17
      simple_knn.cu:17:10: fatal error: cub/cub.cuh: No such file or directory
         17 | #include <cub/cub.cuh>
            |          ^~~~~~~~~~~~~
      compilation terminated.  
      error: command '/home/jinwoo/anaconda3/envs/loop_splat/bin/nvcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simple_knn
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      running bdist_wheel
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:502: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-310
[... 86 lines cut ...]
     copying diff_gaussian_rasterization/__init__.py -> build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      running build_ext
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:414: UserWarning: The detected CUDA version (12.4) has a minor version mismatch with the version that was used to compile PyTorch (12.1). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      /home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/utils/cpp_extension.py:424: UserWarning: There are no /usr/bin/g++ version bounds defined for CUDA version 12.4
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'diff_gaussian_rasterization._C' extension
      creating build/temp.linux-x86_64-cpython-310
      creating build/temp.linux-x86_64-cpython-310/cuda_rasterizer
      /home/jinwoo/anaconda3/envs/loop_splat/bin/nvcc -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/TH -I/home/jinwoo/anaconda3/envs/loop_splat/lib/python3.10/site-packages/torch/include/THC -I/home/jinwoo/anaconda3/envs/loop_splat/include -I/home/jinwoo/anaconda3/envs/loop_splat/include/python3.10 -c cuda_rasterizer/backward.cu -o build/temp.linux-x86_64-cpython-310/cuda_rasterizer/backward.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -I/tmp/pip-req-build-o29ti5x6/third_party/glm/ -DTORCH_API_INCLUDE_EXTENSION_H -D [... cut ...]
      error: command '/home/jinwoo/anaconda3/envs/loop_splat/bin/nvcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff_gaussian_rasterization
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (simple_knn, gaussian_rasterizer, diff_gaussian_rasterization)
                                                                                                      failed

CondaEnvException: Pip failed

Zhu-Liyuan, 2024-09-23:
Thank you for reporting the bug. It seemed that there was a cuda version mismatch in the installation guide. We have updated the installation accordingly. Feel free to git pull the code and follow the new instructions.

closed 2024-09-24 by Zhu-Liyuan

jinxxo-j, 2024-09-24:
Thank you ! 
Your help has been a great assistance