GradientSpaces/LoopSplat #11: Failed to build simple_knn

Hunamic, 2024-10-30:
Hello. Thanks for your great work. 
I have a problem building simple_knn git dependency from requirements.txt(```git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d```). 
I encountered the following error:
**23 | #define __CUDACC__
            |
      <command-line>: note: this is the location of the previous definition
      simple_knn.cu(90): error: identifier "FLT_MAX" is undefined
          me.minn = { FLT_MAX, FLT_MAX, FLT_MAX };
                      ^
      
      simple_knn.cu(154): error: identifier "FLT_MAX" is undefined
         float best[3] = { FLT_MAX, FLT_MAX, FLT_MAX };
                           ^
      
      2 errors detected in the compilation of "simple_knn.cu".
      error: command '/usr/local/cuda/bin/nvcc' failed with exit code 2
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simple_knn
  Running setup.py clean for simple_knn
Failed to build simple_knn
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (simple_knn)**

I will appreciate your support to solve this problem

Zhu-Liyuan, 2024-10-30:
Hi, thank you for your interest. Can you show the complete error output and the environment information of your computer?

closed 2024-10-31 by Zhu-Liyuan

Hunamic, 2024-11-02:
The device and package information:
```
GPU: GeForce RTX 3060
OS: Ubuntu 20.04.6 LTS
Python version: 3.10.15 
CUDA version: V12.6.20
```
I didn't use Conda to install packages as the CUDA toolkit was installed in my OS.
This is the complete error of the code after installing the required libraries:
```
Building wheels for collected packages: simple_knn
  Building wheel for simple_knn (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      running bdist_wheel
      /home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/utils/cpp_extension.py:497: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_ext
      /home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/utils/cpp_extension.py:416: UserWarning: The detected CUDA version (12.6) has a minor version mismatch with the version that was used to compile PyTorch (12.4). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      /home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/utils/cpp_extension.py:426: UserWarning: There are no x86_64-linux-gnu-g++ version bounds defined for CUDA version 12.6
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'simple_knn._C' extension
      creating build/temp.linux-x86_64-cpython-310
      x86_64-linux-gnu-g++ -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/TH -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/include -I/usr/include/python3.10 -c ext.cpp -o build/temp.linux-x86_64-cpython-310/ext.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
      /home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/utils/cpp_extension.py:1964: UserWarning: TORCH_CUDA_ARCH_LIST is not set, all archs for visible cards are included for compilation.
      If this is not desired, please set os.environ['TORCH_CUDA_ARCH_LIST'].
        warnings.warn(
      /usr/local/cuda/bin/nvcc -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/TH -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/lib/python3.10/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/hunamic/PycharmProjects/SLAM/LoopSplat/.venv/include -I/usr/include/python3.10 -c simple_knn.cu -o build/temp.linux-x86_64-cpython-310/simple_knn.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++17
      simple_knn.cu:23: warning: "__CUDACC__" redefined
         23 | #define __CUDACC__
            |
      <command-line>: note: this is the location of the previous definition
      simple_knn.cu(90): error: identifier "FLT_MAX" is undefined
          me.minn = { FLT_MAX, FLT_MAX, FLT_MAX };
                      ^
      
      simple_knn.cu(154): error: identifier "FLT_MAX" is undefined
         float best[3] = { FLT_MAX, FLT_MAX, FLT_MAX };
                           ^
      
      2 errors detected in the compilation of "simple_knn.cu".
      error: command '/usr/local/cuda/bin/nvcc' failed with exit code 2
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simple_knn
  Running setup.py clean for simple_knn
Failed to build simple_knn
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (simple_knn)
```

unique1i, 2024-11-02:
Hi, simple_knn has issue when building with cuda 12.6. It's tested working with cuda 12.1 and 12.4. Please refer to our environment setups.

Hunamic, 2024-11-04:
Thanks. I could resolve the problem of simple_knn by cloning the "https://gitlab.inria.fr/bkerbl/simple-knn" and building from the project setup file in Cuda 12.6.