GradientSpaces/LoopSplat #16: question about the results of Replica

ZoranGong, 2024-12-02:
I found the result of MonoGS in your tab. 1 to be very different from its original paper, especially _off 4_. To ensure it, I run MonoGS in _off 4_ to see how it works but I cannot reproduce the value in your tab. 1. Can you tell me how to get it?

Zhu-Liyuan, 2024-12-02:
We take the number from Tab.2 (sp) in the original paper (https://arxiv.org/abs/2312.06741). 
[image].
I don't understand what you meant by "very different from its original paper" because it's the same as the original paper.

ZoranGong, 2024-12-02:
sorry, my bad. I didn't notice they revised the value in their latest version. I'll try to use the single process they mentioned.

Thanks for your help and quick response

closed 2024-12-02 by Zhu-Liyuan

EH-HU, 2024-12-04:
Thank you for the prompt response. I will close this issue.