GradientSpaces/LoopSplat #17: Question About Frame-to-Model Tracking

EH-HU, 2024-12-03:
Hello,

I have a question about the paper.  
Why does the method optimize the relative camera pose T_(j-1,j) between frames j and j-1 instead of directly optimizing the current camera pose T_j?  
What are the advantages of this approach compared to directly optimizing T_j?  

In relation to this, I also have a question about the code in Tracker.py at line 141:  

init_rel = init_c2w @ np.linalg.inv(last_c2w)  

Does init_rel represent T_(j-1,j) as described in the paper?  
If so, shouldn’t T_(j-1,j) be computed as:  

init_rel = np.linalg.inv(last_c2w) @ init_c2w  

Could you clarify the intent of this code and the mathematical reasoning behind it?  

Thank you!

Zhu-Liyuan, 2024-12-03:
Hi,

1. Your observation is correct. The reason behind this is that the origianl Gaussian Rasterizer only has cuda backward implementations on Gaussian means, rotation, scale and SH. So we couldn't explicitly calculate gradients for the camera pose using CUDA directly with the rasterizer. Instead, we use torch to optimize the relative pose change of Gaussians wrt the previous frame, which is equivalent to directly optimizing current camera pose. 

2. The pose change we get from 1. is the pose change in Gaussians not in the camera. Thus, it is multiplied to the right of the camera pose.

Hope this answers your questions.

EH-HU, 2024-12-04:
Thank you for the prompt response; it has clarified my questions.
However, if explicit gradient calculation for the camera pose becomes feasible, would directly optimizing the pose of frame j be a better approach?

Zhu-Liyuan, 2024-12-04:
In terms of pose accuracy, I don't think so. As I said, optimizing either of them is equivalent to the other one. MonoGS actually implemented explicit gradient calculation in their rasterizer and I tried it in my experiments, which didn't provide obvious improvement. But in term of speed, explicit cuda implementation might be faster than numerical gradient calculated in pytorch (this I didn't try).

EH-HU, 2024-12-04:
Thank you for the prompt response. I will close this issue.

closed 2024-12-04 by EH-HU