GradientSpaces/LoopSplat #18: Failed to get reasonable results on ScanNet

XXXCARREY, 2024-12-05:
Thank you for your amazing work! I encountered a problem while reproducing LoopSplat on ScanNet. During tracking, the camera error rapidly increases, eventually leading to tracking failure. As shown in the picture, the values are still relatively normal in Tracking frame 3, but issues arise in Tracking frame 4. Have you experienced this issue, and how can it be resolved? I have already processed the dataset in advance using `scannet_preprocess.ipynb`.
`
[image]

unique1i, 2024-12-05:
Hi, thanks for the feedback. We haven't noticed the spike of error, can you provide which scannet scene is this?

XXXCARREY, 2024-12-05:
Sorry, I found that only the results on the ScanNet scenes had issues, while the other datasets are normal. Therefore, I generated and processed ScanNet again, and the problem has now been resolved. Thank you very much for your quick response.

closed 2024-12-05 by XXXCARREY