GradientSpaces/LoopSplat #19: Question about overlap ratio

EH-HU, 2024-12-11:
In the process of determining the overlap ratio, is it correct to assume that a point is considered overlapping between two submaps if the nearest neighbor distance is smaller than a threshold?
If so, could the accumulated tracking errors cause the distance between the submaps to increase significantly, resulting in a low overlap ratio even for a valid loop, potentially leading to the loop being misclassified as invalid?

Zhu-Liyuan, 2024-12-11:
This is a good question. The accumulated tracking error could potentially be a source of error. That's why we perform our loop closure when the current submap is finished. By doing so, we make sure that the global tracking error with stay low in the overlapping regions. We haven't seen such case on the four datasets we reported. Online loop closure is very important. We have also tried only doing loop closure after the whole sequence but it led to poor results.

What you described is posiible to happen, but if this happened, we wouldn't know:
- if zero overlap between submaps is caused by invalid loop detection
- or it's caused by drastic errors in tracking
Based on the two observations, we select a relatively loose threshold 0.2 for the overlap ratio. In point cloud registration, with overlap ratio > 0.5 could the two submaps be well registered. Loop detection using both images and 3d geometry can provide a more robust estimate that just using one of them. I have experimented with only using images and it resulted in slightly worse performance. Hope this answers your questions.

EH-HU, 2024-12-11:
[image]
In the supplementary material of the paper, the part where 
T_(p→q) (P) is applied is confusing. Wasn’t the distance between points supposed to be calculated before applying the transformation?

Zhu-Liyuan, 2024-12-11:
Thanks for pointing this out. I also found this in the overleaf and we are going to update this formula in camera-ready version.

closed 2024-12-13 by Zhu-Liyuan