GradientSpaces/LoopSplat #21: Error on Running Code

Anniebhalla16, 2025-03-14:
Hi,
Kuddos to the great work you guys have done. On running LoopSplat on TUM_RGBD data I am getting the following aborted (core dumped) error. Could you help me resolve this?

This is the command I used.

python run_slam.py configs/TUM_RGBD/rgbd_dataset_freiburg1_room.yaml --input_path scripts/data/TUM_RGBD-SLAM/rgbd_dataset_freiburg1_room --output_path output/

[image]

Zhu-Liyuan, 2025-03-18:
Hi,

thank you for your interest in our work and reporting the bug. Could you please share the specs of your computer, as well as the version of torch and cuda? 

thanks

Zhu-Liyuan, 2025-03-27:
Closing due to inactivity. Feel free to reopen in the future. Thanks.

closed 2025-03-27 by Zhu-Liyuan