GradientSpaces/LoopSplat #22: Faiss assertion Error

mah-asd, 2025-04-01:
Hi 
Thanks for publishing the code. 
I'm running on a A5000 Nvidia GPU. and I'm getting this error:

  x.storage().data_ptr() + x.storage_offset() * 4)
Faiss assertion 'err == CUBLAS_STATUS_SUCCESS' failed in void faiss::gpu::runMatrixMult(faiss::gpu::Tensor<float, 2, true>&, bool, faiss::gpu::Tensor<T, 2, true>&, bool, faiss::gpu::Tensor<IndexType, 2, true>&, bool, float, float, cublasHandle_t, cudaStream_t) [with AT = float; BT = float; cublasHandle_t = cublasContext*; cudaStream_t = CUstream_st*] at /project/faiss/faiss/gpu/utils/MatrixMult-inl.cuh:265; details: cublas failed (13): (1024, 3) x (131072, 3)' = (1024, 131072) gemm params m 131072 n 1024 k 3 trA T trB N lda 3 ldb 3 ldc 131072
*** Aborted at 1743548003 (unix time) try "date -d @1743548003" if you are using GNU date ***
PC: @     0x74ec128969fc pthread_kill

Zhu-Liyuan, 2025-04-02:
Hi,

Thank you for reporting this issue. It seems to be a Faiss related issue and we don't have a A5000 to reproduce this error.

Could you check this relevant [issue](https://github.com/GradientSpaces/LoopSplat/issues/5) or look for solutions in the [faiss](https://github.com/facebookresearch/faiss) library?

Best,
Liyuan

mah-asd, 2025-04-02:
Hi, thanks for replying, 
I tried with replacing gpu_index.add(tgt_tensor.cuda()) and gpu_index = faiss.index_cpu_to_gpu(faiss.StandardGpuResources(), 0, cpu_index), still getting the error. 
The error is after running for 370 frames. 

I'll try to look at the faiss library to see if I can resolve this. 

Thanks

Zhu-Liyuan, 2025-04-04:
You can double check your pytorch, cuda and faiss version and see it they match.

closed 2025-04-14 by Zhu-Liyuan

zuoweic, 2025-07-09:
You can try to install faiss-gpu from the source,  download the zip from https://github.com/facebookresearch/faiss/releases , unzip, get into the directory, and then activate your conda environment, then       

cmake -B build . \
    -DFAISS_ENABLE_GPU=ON \
    -DBUILD_TESTING=OFF \
    -DCMAKE_BUILD_TYPE=Release \
    -DCUDAToolkit_ROOT=/usr/local/cuda-12.8 \
    -DPython_EXECUTABLE=/home/c/anaconda3/envs/loopsplatonly/bin/python(you can get this address through "which python")
  then  it works for me.