GradientSpaces/LoopSplat #23: Core Dumped error on run

Anniebhalla16, 2025-04-03:
Hi,

I am reopening this issue regarding core dumped error on running code. I am providing the following information asked [here](https://github.com/GradientSpaces/LoopSplat/issues/21)

This is the command I used.

python run_slam.py configs/TUM_RGBD/rgbd_dataset_freiburg1_room.yaml --input_path scripts/data/TUM_RGBD-SLAM/rgbd_dataset_freiburg1_room --output_path output/

[image]

Cuda version: 12.4
Torch Version: 2.5.1
System specs:  

`Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          46 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   32
  On-line CPU(s) list:    0-31
Vendor ID:                GenuineIntel
  Model name:             13th Gen Intel(R) Core(TM) i9-13900K
    CPU family:           6
    Model:                183
    Thread(s) per core:   2
    Core(s) per socket:   24
    Socket(s):            1
    Stepping:             1
    CPU max MHz:          5800.0000
    CPU min MHz:          800.0000
`

Thanks in Advance
Best
Annie

Zhu-Liyuan, 2025-04-03:
Are you running it on Windows/Linux or MacOS? What GPU are you using?

Anniebhalla16, 2025-04-04:
After installing faiss-gpu like this
pip install faiss-gpu-cu12
I am able to run the code. However I have faiss-gpu latest version 1.10.0 .

Thanks for the help :D

Zhu-Liyuan, 2025-04-04:
Great you solved the problem! It seemed there was a mismatch on cuda versions between your system cuda and faiss you installed before. 

Issue resolved.

closed 2025-04-04 by Zhu-Liyuan

Anniebhalla16, 2025-04-30:
Hi again,

The loop_splat pipeline was running perfectly fine.
However, days later when I go back tot he code and run the same command 
`python run_slam.py configs/TUM_RGBD/rgbd_dataset_freiburg1_room.yaml --input_path scripts/data/TUM_RGBD-SLAM/rgbd_dataset_freiburg1_room --output_path output/`

I get the following error 

[image]

Why does this happen?