GradientSpaces/LoopSplat #24: Why are the transformation for the odometry edges set to identical?

lemonci, 2025-04-29:
From

https://github.com/GradientSpaces/LoopSplat/blob/676e18f950b2de6be39525a613120712b67768bb/src/entities/lc.py#L219C1-L228C82

and

https://github.com/GradientSpaces/LoopSplat/blob/676e18f950b2de6be39525a613120712b67768bb/src/entities/lc.py#L524C1-L526C45

we know that the transformation of all odometry edges were set to be identity matrix.

However, you specifically start a new submap based on the motion heuristic:

https://github.com/GradientSpaces/LoopSplat/blob/676e18f950b2de6be39525a613120712b67768bb/src/entities/gaussian_slam.py#L100C1-L102C50

So may I ask why you don't use the estimated transformation between submaps but the identity matrix for odometry edges transformation?

Zhu-Liyuan, 2025-04-30:
Hi,

thank you for your questions. 

1. "we know that the transformations of all odometry edges were set to be identity matrix." yes this is because in pose graph optimziation (PGO) we only calculate the pose correction to each submap. All the submaps are computed within a global reference frame.
2.  "Why don't use the estimated transformation between submaps" - They are all expressed in the global reference frame so there is no transformation we can extract. Submap pose correction is not equal to relative camera motion in our motion heuristics.

Hope this addresses your questions. 
Best

lemonci, 2025-04-30:
Hi,

I think according to the [documentation of Open3D](https://www.open3d.org/docs/latest/tutorial/geometry/transformation.html), the transformation is just the relative translation and rotation between two submaps. As long as they are under the same coordinate system, it is irrelevant which exact coordinate system they are in.

On the other hand, if you assume the nodes have identity matrices, it means you assume that the agent did not move at all in consecutive submaps, which is contradictory to the fact that you used motion heuristics to start new submaps.

Zhu-Liyuan, 2025-04-30:
The motion heuristic is based on camera motion not the submap motion. The pose of submap does not represent the pose of the agent.

lemonci, 2025-04-30:
Thanks. I do understand we cannot use motion heuristic for submap transformations.
On the other hand, we do have estimated camera to world transformation of each submap. So why we cannot use the transformation of consecutive sub amps calculated based on the estimated camera to world transformation of each?

lemonci, 2025-05-01:
Hi,

Could you keep this issue open issue for a week? My supervisors have different opinions about the transformations of odometry edges and we will have a discussion and get back to you later.

Thanks.

lemonci, 2025-05-10:
Hello,

I am back :) 

What I meant is why you have to use correction for pose graph instead of directly integrating the transformation to Graphslam like what the author of Gaussian-SLAM did in their following work:

https://github.com/VladimirYugay/MAGiC-SLAM/blob/main/src/entities/pose_graph_adapter.py

Zhu-Liyuan, 2025-05-14:
1. "why you have to use correction for pose graph instead of directly integrating the transformation to Graphslam". It was a design choice I followed in Loopy-SLAM. I think the two types of implementation you mentioned are both correct and equivalent. Feel free to implement it yourself.
2. MAGiC-SLAM is a work after LoopSplat and I haven't had time to check their code yet, nor their issue.

lemonci, 2025-05-17:
> 1. "why you have to use correction for pose graph instead of directly integrating the transformation to Graphslam". It was a design choice I followed in Loopy-SLAM. I think the two types of implementation you mentioned are both correct and equivalent. Sorry I am currently busy with other deadlines so I couldn't try it myself. Feel free to implement it yourself.
> 2. MAGiC-SLAM is a work after LoopSplat and I haven't had time to check their code yet, nor their issue.

Thanks. I will directly ask the author of Loopy-SLAM(also the author of MAGiC-SLAM).

closed 2025-05-17 by lemonci