GradientSpaces/LoopSplat #25: Error running LoopSplat

Anniebhalla16, 2025-04-30:
Hi again,

The loop_splat pipeline was running perfectly fine.
However, days later when I go back to the code and run the same command
`python run_slam.py configs/TUM_RGBD/rgbd_dataset_freiburg1_room.yaml --input_path scripts/data/TUM_RGBD-SLAM/rgbd_dataset_freiburg1_room --output_path output/`

I get the following error

[image]

Why does this happen?

Thanks

Zhu-Liyuan, 2025-04-30:
Hi, 

"It was running fine" - which means the environment and code were bug-free back then.
From the image, you have some issues with simple_knn library. My guess is you either have changed the environment of LoopSplat (e.g., you installed some other libraries and changed some existing libraries), or you might have updated the CUDA versions. 

To fix this issue, you could try:

- Reinstalling the simple_knn package with the correct PyTorch/CUDA versions
- Rebuilding the extension from source
- Checking for compatible versions of the dependencies

Anniebhalla16, 2025-05-04:
Thankyou.

I ran loop splat again and I got the following output.
[image]

I'm getting a prompt that says "killed" at 479/494 in reconstruction evaluation. I do not completely understand the reason behind it.

Anniebhalla16, 2025-05-05:
Thank you, it is working again :)

closed 2025-05-05 by Zhu-Liyuan