GradientSpaces/LoopSplat #26: LooSplat on outdoor dataset

Anniebhalla16, 2025-05-07:
Hi,

With high interest in your work, I am keen to knowing whether you have also tried to run loopsplat on outdoor datasets? If yes, do you also have an opensource dataset that can be used by us. And what is the performance of Loopsplat in this scenario (if applicable) .

Thanks in advance
Annie

Zhu-Liyuan, 2025-05-07:
Hi,

I haven't tested it on outdoor dataset yet. In principle LoopSplat should work on outdoor dataset. Be aware that outdoor depth sensors (LiDARs, stereo cameras) might be very different from RGB-D cameras and you need to change some hyperparameters of the system. There have been some systems dedicated for outdoor scenes (https://github.com/PRBonn/PINGS). 

Best,
L

Anniebhalla16, 2025-05-15:
Okay. Could you elaborate a bit more on what hyperparameters should I keep in mind while working with the outdoor dataset?

Zhu-Liyuan, 2025-05-19:
Hi,

You might want to change the range of the depth sensors, the variance and scale of Gaussians, loop closure thresholds and size of the submap.

Zhu-Liyuan, 2025-06-08:
Closing due to inactivity. Feel free to open another issue if you have questions in the future.

closed 2025-06-08 by Zhu-Liyuan