GradientSpaces/LoopSplat #33: Tracking Loss

Anniebhalla16, 2025-07-25:
Hi,
I am currently trying to run loopsplat on my outdoor dataset. I currently dont have a ground truth gps poses, for poses im using an identity matrix.
During tracking I get color_loss 0 and depth loss 0 as well.
It does not seem to be ideal to me. Could you help me understand why this could possibly be ?
I am attaching a screenshot.

[image]

Thanks 
- Annie

Anniebhalla16, 2025-07-30:
HI,
I would also like to understand how to tune a few hyper parameters for my outdoor dataset. 
I noticed that a mesh .ply file is generated during the evaluation process only and also not before evaluation. Is there a particular reason for that?

Thanks in Advance

Zhu-Liyuan, 2025-07-31:
- For tracking loss, can you visualize the camera pose and intermediate 3d splats to check if you have the correct Gaussian intialization and the coordinate system is correctly defined?
- For outdoor dataset, I haven't tested it out either so I couldn't provide much useful advice.

closed 2025-08-15 by Zhu-Liyuan