GradientSpaces/LoopSplat #34: Something wrong in ScanNet 0059

BW-Zhang05, 2025-08-13:
Mapping frame 1786
Number of added points:  120518
Gaussian model size 120518
Optimization time:  0.7509114742279053
PSNR this frame: 21.17544937133789
Loaded trained exposures paramters for scene scene0059_00
Starting evaluation...🍺
Running trajectory evaluation...
ATE-RMSE: 155707.76 cm, ATE-RMSE (aligned): 120667.79 cm
Running reconstruction evaluation...
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 36/36 [00:11<00:00,  3.08it/s]
Mesh cleaning (before/after), vertices: 27416/0, faces: 37078/0
[Open3D WARNING] Write PLY failed: mesh has 0 vertices.
Reconstructed mesh saved to output/ScanNet/scene0059/mesh/cleaned_mesh.ply
Running global map evaluation...
RPly: Unable to open file
[Open3D WARNING] Read PLY failed: unable to open file: output/ScanNet/scene0059/mesh/cleaned_mesh.ply
Number of points at initialisation :  0
Refinement:   0%|                                                                                                                                               | 0/30000 [00:00<?, ?it/s]
Could not run global map eval
Traceback (most recent call last):
  File "/root/LoopSplat/src/evaluation/evaluator.py", line 351, in run
    self.run_global_map_eval()
  File "/root/LoopSplat/src/evaluation/evaluator.py", line 226, in run_global_map_eval
    refined_merged_gaussian_model = refine_global_map(merged_cloud, training_frames, 30000, export_refine_mesh=False,
  File "/root/LoopSplat/src/evaluation/evaluate_merged_map.py", line 128, in refine_global_map
    training_frame = next(training_frames)
  File "/miniconda3/envs/loopsplat/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 630, in __next__
    data = self._next_data()
  File "/miniconda3/envs/loopsplat/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 674, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "/miniconda3/envs/loopsplat/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 51, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/miniconda3/envs/loopsplat/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 51, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/LoopSplat/src/evaluation/evaluate_merged_map.py", line 48, in __getitem__
    "render_settings": get_render_settings(
  File "/root/LoopSplat/src/utils/utils.py", line 94, in get_render_settings
    cam_center = torch.inverse(w2c)[:3, 3]
RuntimeError: CUDA error: invalid configuration argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

All done.✨

**The ATE is very poor， and  mesh and .ply is not generate. Whats the problem**

Zhu-Liyuan, 2025-08-13:
Hi, 

Did you follow our instructions in README to process ScanNet dataset? 

`Pay attention! There are some frames in ScanNet with inf poses, we filter them out using the jupyter notebook scripts/scannet_preprocess.ipynb. Please change the path to your ScanNet data and run the cells.`

Zhu-Liyuan, 2025-08-13:
This is very unusual. I have tested many times on all Scannet scenes we reported in the paper, and the performance on Scannet was quite stable. Did you try other scenes?

closed 2025-08-15 by BW-Zhang05